RobTillaart/AD524X #6: Constantly resetting to mid point

3DALT, 2021-01-27:
I'm attempting to use this library with a Teensy 4.0 and the potentiometer keeps reading the midpoint value. Simply put, I can't seem to get this to work. It seems to be a communication error but I've noticed the read commands don't actually read from the AD524X but simply return what you last set the value as.

RobTillaart, 2021-01-27:
Thanks for pointing out,
I did not see this before, 
- can you post a minimal sketch that shows the problem. 
- have you checked wires?  As it sounds like reset it could be some short circuit. Might even be in a breadboard.
- have to check if I have hardware around to verify.

Will be back later today or tomorrow

RobTillaart, 2021-01-27:
@3DALT 

### Analysis
- The read command does indeed return the **_lastValue** set as that is much faster.
- The write command it sets the **_lastValue** 
- The initial value from the constructor is 127 which is midpoint,

=> So the write command seems to fail.

### Test if it is connected.

Can you test this sketch and send me the output?
```cpp
//
//    FILE: AD524X_isCOnnected.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.0
// PURPOSE: AD524X demo program
//    DATE: 2021-01-27
//     URL: https://github.com/RobTillaart/AD524X
//

#include "AD524X.h"

AD524X AD01(0x2C);     //  AD0 & AD1 == GND

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.println(AD524X_VERSION);

  Wire.begin();
  Wire.setClock(400000);

  bool b = AD01.begin();
  Serial.println(b?"true":"false");
  Serial.println(AD01.isConnected());

  Serial.println("\nDone...");
}

void loop()
{
}

// -- END OF FILE --

```

3DALT, 2021-01-27:
[image]
I've attached my circuit. Fairly simple, just i2c connected to teensy through a level shifter. I'm using the pot in the voltage divider mode for simple testing. 

Response from your new code is:
```
C:\Users\aamar\Documents\Arduino\AD524X_isConnected\AD524X_isConnected.ino

0.3.1

false

0

Done...

```

Could the issue be related to needing an R/W bit at the end of your address byte?

I'll try using a different breadboard and double-check my wiring. 

Thank you so much for your quick reply and help.

RobTillaart, 2021-01-27:
The output states that the teensy cannot see the AD524X.
So it can only return default.

The level converter is good

Do you have long wires?
Do you have pull up resistors? 
On both sides of the level converter?

Pull ups are needed to get a nice square signal 
- perfectly explained there ==>  http://www.gammon.com.au/forum/?id=10896&reply=5#reply5

RobTillaart, 2021-01-27:
Do you have connected the GND of 3.3V and 5V?
Both signals should have a common reference.

3DALT, 2021-01-28:
I'm not entirely sure what happened because I was able to communicate yesterday but was not able to this morning. After messing around with a scope I was finally able to get to communicating again and pullup resistors definitely help with the signal consistency. Everything seems to work fine now. 

 Thank you again for your help. Much appreciated.

closed 2021-01-28 by 3DALT

RobTillaart, 2021-01-28:
Welcome
Added the example above to the repo