RobTillaart/AD524X #8: identity request

fotosettore, 2021-11-12:
hi 
is it possible in some way (ex. a read instruction) to know the type of IC and its pots value ?

RobTillaart, 2021-11-12:
mmm, good question, you want to determine runtime if you have an 41 or 42?

I am not aware of any register that gives this information, nor a register that states the pots type.
(there might be hidden registers)

**thinking out loud**
That would mean you instantiate an object of type AD524X or AD5242 so you can access both ports.
Then write 0 to port 2 , read back from register , write 1, read back from register .
If you read back the same value as you wrote, it might be an indication that it is indeed a 42 with 2 ports.
At least if it fails it sure is a 41. 

That is how it could be done. 

Did you check the datasheet wrt this question?

RobTillaart, 2021-11-16:
Page 5 datasheet  

[image]

There is no information about reading the resistance of the device.

RobTillaart, 2021-11-16:
so in code it would look like this. Not tested 

```cpp
uint8_t detectType()
{
  AD.write(1, 0);  // set value RDAC 2 to zero
  if (AD.readRegister() != 0) return 41;
  AD.write(1, 42);  // set value RDAC 2 to zero
  if (AD.readRegister() != 42) return 41;
  return 42;
}
```

Drawback of this routine is that it will affect the value of the device, which will affect the project in which it is used.
As this is a serious negative side effect I will **not** add this to the library.

closed 2021-11-16 by RobTillaart

fotosettore, 2021-11-17:
hi
thanks for your answer!
i'm very wondered that Analog Device did not put inside infos about it.
Two bytes would have been enough, even one to be honest !!!!
a hex num and a conversion table  : 0x00 ? ok is a AD5252BRUZ100 ---- 0x01 = ok, is a BRUZ200 ...... 0xFA ? ok, is a 5280 bruz etc etc
I add another gap : ADAD525x series has a two type of pots
The 5253 is a 64 steps and 5254 is a 256 steps
Nothing say to me the type (you understand that depending of resolution you must change the mapping value)
really really a pity ... 
unfortunately no competitor factory about I2C digital pots :-(

RobTillaart, 2021-11-17:
Yes 3 bytes would be ideal 

1 byte for # ports
1 byte for resistance   (maybe even per port if these could be different)
1 byte for # steps - formula   2^n

more digital pots are there from Maxim and Microchip

(in dutch but they probably have italian site too)
https://nl.farnell.com/w/c/semiconductors-ics/digital-potentiometers/prl/resultaten?control-interface=i2c&st=digital%20potentiometer

fotosettore, 2021-11-17:
oohhh wonderful !!
last year I search for any competitor but i found only SPI port products
your info has opened up a world for me!!:-))
really many thanks

RobTillaart, 2021-11-17:
you're welcome