RobTillaart/AD524X #10: Unable to see results as expected.

EARTKEY, 2022-02-27:
Hi Everyone, 
I'm looking for guidance to know what exactly the reason that my circuit is not working as expected. 

1. I have connected to AD5242 IC as per the datasheet guidance and integrated with ATTINY402 using I2C bus. As always pull-up registers and other connections are taken care. 
2. I currently have followed only RobTillaart AD524X library and its example codes. Initially I tested with "isConnected" example code and it responded true ('1') on serial monitor. Later I ever tested with logic outputs "O1 & O2" and they are working absolutely fine.

- Now the problem is, Even after RDAC1 & RDAC2 registers are being updated with "200" value, there is no change in voltage drop at W1 & W2 with respect to B1 & B2 (Ground connected). I did some survey online and tried all possible way, but unable to find a solution to this. :( 
- Even I tried reading RDAC register back for validation, surprisingly register hold "200" in it. However, A1, W1 & B1 terminals are not at all changing. 

PFA schematics and code for your reference. 
[image]
[image]

I'm very happy if anyone over GitHub can help me to solve this problem. 

Thanks in advance, 
Vaibhav Sugandhi

RobTillaart, 2022-02-27:
This is more a question for the Arduino forum as it is not a problem of the library.

If you just measure the resistance A1-W1, and change the position programmatically, does it change?

RobTillaart, 2022-02-27:
How is the SHDN line connected? is it floating?

[image]

RobTillaart, 2022-02-27:
What is the voltage used VBAT?

EARTKEY, 2022-02-28:
Hi Rob, 

1. It is 1M digital Pot, I can see it working as normal voltage divider network. But Resistance is not as expected in OFF state. And yes, I2C is working fine with respect to O1 & O2. 
2. As shown in the schematics. I have pulledup SHDN pin to VCC and connected to GPIO of Attiny402. I'm using this to shutdown the digipot when needed.
3. As mentioned in the schematics, VBAT is 3V coin cell. But right now I'm using 3.3V in Arduino Nano. 

Here is some important update from my trial and error, 
The voltage at A1 is 3.3V & B1 is floating. At this state, W1 will be center tap point, which makes whole circuit as a series resistor. After a lot of try, even replacing the IC AD5242 I found that when I write 255 value to the RDAC1, then W1 is showing 2V. No change in Circuit diagram, it is as mentioned in above schematics only.

RobTillaart, 2022-02-28:
You could connect B1 to W1 to prevent floating, it should not change the R of W1A1

RobTillaart, 2022-03-12:
@EARTKEY 
Any new insights ...?

What are the values of R6, R7 and R8 ?

RobTillaart, 2022-03-27:
As there is no answer I assume the issue is solved.

closed 2022-03-27 by RobTillaart