RobTillaart/AD524X #11: AD5245 supported

street-grease-coder, 2022-07-30:
Hi, in title implicitly states AD5245 is supported, but did not see it written - is this likely to work? AD5245 is one of the top 5 digital potentiometers supported by jlcpcb assembly service currently

RobTillaart, 2022-07-31:
Thanks for the question, 
I need to check the datasheet to see if the 5245 will work with this library. Might take a few days to verify as I have not the hardware to test.

RobTillaart, 2022-07-31:
Had a quick look this morning and it looks like the library could work if addressed as an AD5241

Main difference
- the AD5242/41 have O1 and O2 lines (and thus commands) which are meaningless for the AD5245.
- the AD5242 has two RDAC's - won't be needed.

Think it is well possible to make a AD5245 specific library pretty fast by stripping some code.
At a later stage it could be merged into the AD524X class hierarchy, currently the AD524X base class has to much functionality to derive AD5245 from it. It could be the other way around.

I will create an AD5245 repository to start experimenting asap

RobTillaart, 2022-07-31:
@street-grease-coder 

Created a repo and develop branch here - https://github.com/RobTillaart/AD5245/tree/develop

Please verify if this version works for you.

RobTillaart, 2022-08-03:
@street-grease-coder 
Released 0.1.0 version of ttps://github.com/RobTillaart/AD5245

Please verify if this version works for you. (I have no hardware)
Please follow up issues there.

Hs-Cheng, 2023-08-02:
@RobTillaart @street-grease-coder 
Hello,
I would like to ask if the current library supports AD5245. I clicked into the repo above for AD5245, but the link is broken.
I would also like to ask if the current library supports AD5246. AD5246 has less pins than AD5245 and it seems to be more friendly to me.
@RobTillaart could you please check the compatibility to AD5246? Thank you very much for your help.

RobTillaart, 2023-08-02:
@Hs-Cheng 
Try https://github.com/RobTillaart/AD5245

RobTillaart, 2023-08-02:
@Hs-Cheng 
For the 5246 question i will come back later.

RobTillaart, 2023-08-02:
@Hs-Cheng 
Derived from the AD5245 potentiometer library I derived a dedicated library for the AD5246 rheostat
See - https://github.com/RobTillaart/AD5246

If there are issues with it, please open one in that repository.

closed 2023-08-02 by RobTillaart