RobTillaart/AD524X #16: AD524X with Adafruit RP2040 Feather

tstahov, 2023-01-20:
Thanks for creating a great library!

I am trying to use the library on an Adafruit RP2040 Feather. 
There are two I2C channels on this board. I initialize the library for the second channel (Wire1) as that is the one the AD5242 is connected to:

AD5242 AD01(0x2C, &Wire1);

The setup code indicates that the AD5242 is found. More specifically, this code:
      b = AD01.begin();
      Serial.println(b ? "Begin: true" : "Begin: false");
      Serial.println(AD01.isConnected());
produces  this output:
      Begin: true
      1
 
However, when I try to write to the AD5242 there are errors: 
      int a = AD01.write(100);
      int b = AD01.write(100 );
      char buff[256];
      sprintf(buff,"Write 1: %d\tWrite 2: %d\n", a, b);
      Serial.println(buff);
The output is: 
      Write 1: 4	Write 2: 4
      
From the Arduino documentation for Wire, I believe that a return value of 4 means "other error"
      
I know that the hardware works properly as everything works correctly when I used an ESP32 instead of the RP2040. My board is designed as a feather wing so that I can simply swap feather processors. (For the ESP32 I initialize with AD5242 AD01(0x2C); )  

Any suggestions?

RobTillaart, 2023-01-20:
I have no free RP2040 for testing (need to order some) so first thing I would check is with an I2C scanner if the device can be reached without problems. (apparently the ESP32 can).

e.g. https://github.com/RobTillaart/I2C_SCANNER

RobTillaart, 2023-01-20:
return value of **write()** is return value of **send()** which is indeed the return value of the **Wire.endTransmission()**
```cpp
uint8_t AD524X::write(const uint8_t rdac, const uint8_t value)
{
  if (rdac >= _pmCount) return AD524X_ERROR;

  uint8_t cmd = (rdac == 0) ? AD524X_RDAC0 : AD524X_RDAC1;
  // apply the output lines
  cmd = cmd | _O1 | _O2;
  _lastValue[rdac] = value;
  return send(cmd, value);
}

...

uint8_t AD524X::send(const uint8_t cmd, const uint8_t value)
{
  Wire.beginTransmission(_address);
  Wire.write(cmd);
  Wire.write(value);
  return Wire.endTransmission();
}
```

RobTillaart, 2023-01-20:
```cpp
uint8_t AD524X::write(const uint8_t rdac, const uint8_t value)
```

Note that **write()** needs 2 parameters while your code provides only 1.

Wondering why the ESP32 did work ?

Note for myself, consider adding **write(value)** with default PM == 0 ?

RobTillaart, 2023-01-20:
looking into the library code of Wire of the RP2040 a return value of 4 could indicate 2 things

```cpp
// Errors:
//  0 : Success
//  1 : Data too long
//  2 : NACK on transmit of address
//  3 : NACK on transmit of data
//  4 : Other error
uint8_t TwoWire::endTransmission(bool stopBit) {
    if (!_running || !_txBegun) {
        return 4;   <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
    }
    _txBegun = false;
    if (!_buffLen) {
        // Special-case 0-len writes which are used for I2C probing
        return _probe(_addr, _sda, _scl, _clkHz) ? 0 : 2;
    } else {
        auto len = _buffLen;
        auto ret = i2c_write_blocking_until(_i2c, _addr, _buff, _buffLen, !stopBit, make_timeout_time_ms(_timeout));
        _buffLen = 0;
        return (ret == len) ? 0 : 4;
    }
}
```

tstahov, 2023-01-20:
I had previously tried the Adafruit testbed library and obtained this result: 

Adafruit I2C Scanner
Default port (Wire) I2C scan: 
Secondary port (Wire1) I2C scan: 0x2C,

RobTillaart, 2023-01-20:
> I had previously tried the Adafruit testbed library and obtained this result:
> 
> Adafruit I2C Scanner Default port (Wire) I2C scan: Secondary port (Wire1) I2C scan: 0x2C,

Looks OK so the device can be seen.
think you need to add the prints in the Wire library to see which code path fails.

tstahov, 2023-01-20:
I ran I2C_scanner_simple and the device was not found. Perhaps the scanner is checking only the first I2C port. 

C:\Users\stahov\Documents\Arduino\libraries\I2C_SCANNER\examples\I2C_scanner_simple\I2C_scanner_simple.ino
I2C_SCANNER_LIB_VERSION: 0.1.3

-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-	
-	-	-	-	-	-	-	-

RobTillaart, 2023-01-20:
> Perhaps the scanner is checking only the first I2C port.

yes, try - I2C_scanner_getWirePortCount.ino

tstahov, 2023-01-20:
Thanks! It's 2:00 am so I will add the prints in the Wire library in the morning.

tstahov, 2023-01-20:
I added print statements in Wire.cpp and found that path B is used

RobTillaart, 2023-01-20:
So not all bytes are sent apparently.
This is a liw level Wire library problem.
The first thing to print now is that timeout variable, as that can be the cause.
Also try to print the number of bytes that are send
Just to get detailed insight where it fails.

tstahov, 2023-01-20:
It looks like this may be a problem with I2C on the RP2040 board. I will use a logic analyzer to try to diagnose this.

RobTillaart, 2023-01-21:
followed the calls goes down to the function **i2c_write_blocking_internal(i2c_inst_t \*i2c, uint8_t addr, const uint8_t \*src, size_t len, bool nostop, check_timeout_fn timeout_check, struct timeout_state \*ts)** that seems to do the work. 

This is a low level function that sends one byte at the time.  checks for timeout after every byte.
Not trivial to analyze but it could help you find where things go wrong. 

----

Other thought
Do you have a different I2C device? if that also fails to work you might have the wrong board selected?
Some boards do not work with the Arduino MBED 2040 See https://github.com/RobTillaart/MCP4725/pull/21

tstahov, 2023-01-21:
I tried a Pi Pico instead of the Adafruit RP2040 Feather. The code runs properly on the Pico but not on the Feather. Thus, either this particular Feather is defective or there is something wrong with the configuration files for the Feather version of the RP2040 board. I will order another Feather to figure out which it is. 

Thanks again for all of the help.

RobTillaart, 2023-01-22:
Maybe the Feather has specific pins for I2C?
Once I had an IC which had swapped to names of two pins on the break out board. 
Took weeks to come up with that idea and minutes to confirm.

Apparently the cause is really outside my library. Although I am interested in the cause, I will not investigate further. 
If you have new insights please post them, as these might be interesting for someone else in the future.

RobTillaart, 2023-01-29:
@tstahov 
As the issue is not caused by the library I close the issue. Feel free to open it again if new insights are made.

closed 2023-01-29 by RobTillaart