RobTillaart/AD524X #18: Problem when not using default Wire channel

tstahov, 2023-02-02:
I have finally gotten to the bottom of why the library doesn't work on an Adafruit Feather RP2040! The problem is here:

```cpp
uint8_t AD524X::send(const uint8_t cmd, const uint8_t value)
{
  Wire.beginTransmission(_address);
  Wire.write(cmd);
  Wire.write(value);
  return Wire.endTransmission();
}
```
This code uses the default Wire channel rather then the one specified during initialization. Changing the code to this, fixed the problem: 

```cpp
uint8_t AD524X::send(const uint8_t cmd, const uint8_t value)
{
  _wire->beginTransmission(_address);
  _wire->write(cmd);
  _wire->write(value);
  return _wire->endTransmission();
}
```

Although I am not using it, I believe this: 

```cpp
uint8_t AD524X::readBackRegister()
{
  Wire.beginTransmission(_address);
  Wire.endTransmission();
  Wire.requestFrom(_address, (uint8_t)1);
  return Wire.read();
}
```

should be changed to this: 

```cpp
uint8_t AD524X::readBackRegister()
{
  _wire->beginTransmission(_address);
  _wire->endTransmission();
  _wire->requestFrom(_address, (uint8_t)1);
  return _wire->read();
}
```

Thanks again for all the hard work in producing this library.

RobTillaart, 2023-02-02:
Thanks for reporting, I am quite busy and think it will be tomorrow before I can dive into it.

RobTillaart, 2023-02-02:
Your analysis indeed shows it is a big BUG (with capitals) and shows I must test more. 
(so little time, so much fun to do)

tstahov, 2023-02-02:
I'm glad it was an easy fix.

RobTillaart, 2023-02-03:
Fixed in version 0.4.0

commit dff167a5 (2023-02-03): fix #18 Wire2 bug (#19)
closed 2023-02-03 by RobTillaart