RobTillaart/AD524X #22: AD5263 support possible?

pe1rmg, 2023-10-09:
Hi Rob,

Would it be possible to include support for the AD5263?
I am specifically looking for an Arduino library which supports a 4 channel, 256 step digital potentiometer, with I2C interface.... (It needs to be compatible with an RP2040 microcontroller, but I do not see any issue with the libraries of you I have been using so far, so I expect that will be no problem)

If needed, I am willing to sponsor the development of a library. :-)  (In that case please contact me at my username AT gmail)

Thanks! 
Paul

RobTillaart, 2023-10-09:
Thanks for the issue. 
Such a library should be possible, need to the datasheet (download), a way to test (you?) and some time.

Not seen the datasheet, I assume it is a straightforward device so it should take 4-8 hrs to set up the things incl build infra and initial documentation.
Will download datasheet later today to see if there are surprises.

Do you need it fast?

RobTillaart, 2023-10-09:
- downloaded datasheet
- created a placeholder => https://github.com/RobTillaart/AD5263

I move this issue to that repository.