RobTillaart/AD524X #23: I2C address in table

motoGiS, 2023-11-29:
Hello! Is the data in the "I2C address" table correct? AD0 and AD1 do not need to be swapped? In the datasheet AD1 is the high bit of the address.

RobTillaart, 2023-11-29:
Thanks for the issue, 
You are referring to the table in the **readme.md** I assume..
Will look into it a.s.a.p.
Which version of the datasheet do you have? (a link is preferred)

RobTillaart, 2023-11-29:
#### Analysis
AD5241, AD5242 datasheet REV C

AD0 is  bit 0
AD1 is bit 1

[image]

Readme.md

[image]

#### Conclusion
Yes you're right, will make a PR  later this week.
(there is a breaking change coming to improve the support for more boards. Need to redo the constructor / begin interface)

motoGiS, 2023-11-29:
I didn't have time, you found it yourself :)

RobTillaart, 2023-11-29:
Any other remarks with respect to the library that need fix or changed?
(I might be able to include it in the same PR).

RobTillaart, 2023-11-29:
Develop branch created and PR build is running.

RobTillaart, 2023-11-29:
> I will be testing the library in my project. Thank you for your efforts.

Would be great if you can test the develop branch in your project,
Appeciated!

commit b9eb3994 (2023-12-05): remove ESP32 specific begin() (#24)

- remove ESP32 specific **begin()**
- add (experimental) support or AD5280 / AD5282.
- fix #23 in readme
- update and add examples.
closed 2023-12-05 by RobTillaart