RobTillaart/AD524X #26: Wiring multiple  AD524x

thechordmaster, 2024-11-04:
Trying to wire multiple AD524x digipots in a cascade. How can I connect three of these to an arduino portenta via an i2c multiplexer please. Possible to share the ideal wiring?
Thanks

RobTillaart, 2024-11-04:
basic cascade to vary voltage connects 

- all B's to GND
- A of top level to 5V (or 3V3 whatever)
- wiper W  of one level to the A of next leve;
- last wiper W is final output

```
  5V ----A[  1000 K  ]B-----  GND
              W
              |
    +---------+
    |
    +----A[   100 K  ]B-----  GND
              W
              |
    +---------+
    |
    +----A[   10 K   ]B-----  GND
              W
              |
              |
              o   between 10 and 1000 K steps of 10 K/ 256
```

A pure resistor cascade can also looks like this.

```
     ----A[  1000K  ]B----+----A[  100 K  ]B----+----A[  10 K  ]B----+---o
               W          |         W           |         W          |
               |          |         |           |         |          |
               +----------+         +-----------+         +----------+
```

RobTillaart, 2024-11-04:
Depending on which AD524x you have, some have two address lines so you can set 4 non conflicting addresses on the I2C bus.

this video shows the basics of tca9548 multiplexer

https://www.youtube.com/watch?v=ZsEXHZW-USM

RobTillaart, 2024-11-05:
Another pure resistor cascade leaves the B pin free so a few wires less.
(again W = Wiper)

```
     ----A[  1000K  ]B    +----A[  100 K  ]B    +----A[  10 K  ]B    +---o
               W          |         W           |         W          |
               |          |         |           |         |          |
               +----------+         +-----------+         +----------+
```

----

### Zero switch

An additional option is to connect the first A with the last B with "zero" switch to get 0 Ohm.
This would solve the Wiper Resistance which is 60- 120 Ohm for the AD5422 (datasheet P3).
Note with three wipers in a cascade the Wiper Resistance adds up to 200-360 Ohm,
Optionally you can place a zero switch over every potentiometer.
Drawback is extra components, control lines and code.

```
                            "zero" switch
      +--------------------------o \o--------------------------------+
      |                                                              |
      |                                                              |
   ---+--A[  1000K  ]B    +----A[  100 K  ]B    +----A[  10 K  ]B    +---o
               W          |         W           |         W          |
               |          |         |           |         |          |
               +----------+         +-----------+         +----------+
```
```

RobTillaart, 2024-11-05:
@thechordmaster 

### Range voltage divider

The following setup has two potmeters, left and right, to define a range.
- X can be set between 0.0 and 5.0 Volt and 
- Y can be set between 0.0 and 5.0 volt.

The third range potentiometer has 256 positions within the defined range range.
E.g if left is set to 2.3 Volt and right is set to 2.5 Volt, 
the range potentiometer divides this 0.2 volt in 256 steps of about a single millivolt (in theory).

```
         left limit pot           right limit pot

  GND ----A[  100K  ]B----  5V  ----A[  100K  ]B---- GND
               W                         W
               |                         |
               |        range pot        |
               |                         |
             X +------A[  100 K  ]B------+ Y
                           W
                           |
                           |
                           o   voltage between (0..Vcc)

```

RobTillaart, 2024-11-13:
the above schema show how to construct coarse / fine controls with multiple potentiometers.
There are other designs possible of course, and it is important to "play" with these designs to understand how they differ.
Then you can see which design matches your ideal / requirements.

More detailed explanations => goto https://electronics.stackexchange.com/