RobTillaart/AD524X #27: cascading two UP/DN and one i2c digipot possible?

thechordmaster, 2024-11-11:
Hi.. I am looking to cascade 1 ad5242 1mohm digipot and 2 ds1804 100k (https://www.analog.com/media/en/technical-documentation/data-sheets/DS1804.pdf).

is this doable?

additionally can 20 such sets be achieved using shift registers and i2c multiplexers controlled by a single arduino?

to paraphrase. 20 cascade sets controlled by a single arduino. Each set contains 1 i2c ad5242 , 1 ds1804-010, and 1 ds1804-100.

RobTillaart, 2024-11-11:
> is this doable?

See no reason why not, 

> additionally can 20 such sets be achieved using shift registers and i2c multiplexers controlled by a single arduino?

_Note: I have no experience with the DS1804_

**Q: which single arduino?**

The I2C is simple math, 
for 20 ad5242 you need 1 i2C mux as the AD5242 allows 4 addresses. 
1 MUX would be able to support 8 x 4 addresses = 32 devices.

The shift registers is a design choice.  

40 DS1804's need  3 pins(?) per device so you need 120 IO lines, that is 15 shift registers  that is quite a bit but doable.

Better connect all UP and DOWN lines to 2 pins and control the 40 CS lines with 3 PCF8575's or 6 shift registers.

Very well doable, 
Use a separate power supply (or maybe even more) as all these devices together use a lot of power.

----

Should you not also have a master volume control? 
A unit 21, or is it included?

RobTillaart, 2024-11-11:
@thechordmaster 

Created a minimal library for the DS1804 based upon the datasheet.
Can be used to bring the build in EEPROM (start up value) into a defined state..

- https://github.com/RobTillaart/DS1804

Not yet released but I expect that within an hour or so.

thechordmaster, 2024-11-12:
thank you.. would it be possible to show a schematic of 2 sets of these cascades (1 ad5242 + 1 ds1804-010 and 1 ds1804-100 plus connected to a portenta? I will try and get it to work with a GUI and report back with my results.
also could you help out with a code where-in the different positions of the digipots are automated every 1 second and the serial monitor prints out the value? maybe I can map this out to the respective sliders in the GUI.

RobTillaart, 2024-11-13:
Schema to control 4 DS1804's
- share U/D pin
- share INC pin
- have unique CS pins

```
              +----------+                           +----------+
              |      INC |-----------------------+---| INC      |
              |          |                       |   |          |
              |      U/D |--------------------+------| U/D      |
              |          |                    |  |   |          |
              |       CS |-----+  +------------------| CS       |
              |          |     |  |           |  |   |          |
              |  DS1804  |     |  |           |  |   |  DS1804  |
              +----------+     |  |           |  |   +----------+
                               |  |           |  |
                               |  |           |  | 
              +----------+     |  |           |  |   +----------+
              |      INC |-----------------------+---| INC      |
              |          |     |  |           |  |   |          |
              |      U/D |--------------------+------| U/D      |
              |          |     |  |           |  |   |          |
              |       CS |--+  |  |  +---------------| CS       |
              |          |  |  |  |  |        |  |   |          |
              |  DS1804  |  |  |  |  |        |  |   |  DS1804  |
              +----------+  |  |  |  |        |  |   +----------+
                            |  |  |  |        |  |
                            |  |  |  |        |  |
                            |  |  |  |        |  |
                            |  |  |  |        |  |
              +-------------------------------------------------+ 
              |             0  1  2  3        u  i              | 
              |                                                 | 
              |                                                 | 
              |                                                 | 
              |                                                 | 
              |                     processor                   | 
              +-------------------------------------------------+ 
```

in the code 
- select the Up or DOWN mode with pin u
- select the device with pin 0, 1 ,2,3, .... n 
- set a HIGH LOW pulse train on pin i  to move the number of steps.

It is possible to select multiple devices and give them a number of pulses so they go up/down in parallel.
- think left and right channel, 
- or set all of them to zero
- more complex patterns to increase parallelism (performance) but this is more difficult coding wise.

thechordmaster, 2024-11-13:
Thanks! Could you kindly share the above schematic with two ad5242x also included? I’m sorry I’m very new to this and my work requires these also to be interfaced with the ds1804

RobTillaart, 2024-11-13:
The AD542x's both just connect to the I2C bus. 

- All connect SDA with processor SDA
- All connect SCL with processor SCL
- All connect GND with processor GND
- power supply is probably separate if you have more than a few.

There is a 3K3 pullup between SDA and Vcc
There is a 3K3 pullup between SCL AND Vcc

That's all

You should buy this book to learn the basics about electronics, you do not need all depth just to get the basics you need

https://www.amazon.com/-/en/Paul-Scherz/dp/1259587541

RobTillaart, 2024-11-18:
Assuming this has answered in enough detail I close this issue.
Feel free to reopen if needed.

thechordmaster, 2024-11-20:
[
[image]
](url)

Hi is this wiring correct for my requirement for one module?

RobTillaart, 2024-11-20:
The lines you draw are correct however some lines are missing

1. from AD5242->SDA to PortentaH7->SDA
2. from DS1804->INC to PortentaH7->GPIOx  (shared by both, to send the increment pulses)
3. from DS1804->UD to PortentaH7->GPIOy (shared by both, to select up / down)
4. PortentaH7 needs a VCC

thechordmaster, 2024-11-20:
[image]
I have attempted to hand draw the entire circuit including shift registers and i2c multiplexers to illustrate the expected requirement. Kindly ignore the badly handing Vcc and GND. Just needed the clarity on whether this circuit would work.

RobTillaart, 2024-11-20:
Looks like you have all control lines, but please 

1. start with one DS1804 until you can control it and understand how to connect the device.
2. add a second DS1804 until you can control both devices
3. Then add the AD5242 until you know how to control that device too.
4. Then add the 74HC595 to control the two DS1804's 
5. Finally add the TCA9548 to multiplex the I2C bus 

You must understand every step to build up your knowledge of these devices and how they behave.
You will need this know how.
Although the schematic has just 5 components it is hard to debug if you do not know how each individual component behaves.
Let me say, this strategy has saved me a lot of time especially for debugging.

RobTillaart, 2024-11-21:
@thechordmaster 

Came across this device today (in fact last night).
The AD5231 is a **1024** position digital potentiometer 10 kΩ, 50 kΩ, and 100 kΩ terminal resistance
The AD5235 is a dual version of the 31
Might do 100 kΩ in about 100 Ω steps.

Some related datasheets.

https://www.analog.com/media/en/technical-documentation/data-sheets/ad5231.pdf
https://www.analog.com/media/en/technical-documentation/data-sheets/ad5235.pdf
https://www.analog.com/media/en/technical-documentation/data-sheets/AD5175.pdf
https://www.analog.com/media/en/technical-documentation/data-sheets/ad5291_5292.pdf

RobTillaart, 2024-11-22:
@thechordmaster 

Under construction - https://github.com/RobTillaart/AD523X
to support the AD5231 and AD5235. 
Imagine putting the two pots of the AD5235 in series, one would get a digipot with 2000 steps.

thechordmaster, 2024-11-23:
Wow this would simplify so many things!
I require a range of 10ohms to 1500kohms. In increments of 10ohms . in this case, stringing 6 of these 250kohms would help me achieve this? So for 20 channels, 20 sets of 6 such cascades?

RobTillaart, 2024-11-23:
Dont know if these have that resolution of 10 ohm in practice. That is to find out by prototype

RobTillaart, 2024-11-23:
released 0.1.0 - https://github.com/RobTillaart/AD523X
not tested with hardware yet.

> I require a range of 10ohms to 1500kohms. In increments of 10ohms

Be aware this is an extreme high precision, more than 1 in a million. 

Have you studied the datasheets?
- what is the accuracy per step? the accuracy of the 1M might be 1K at best (that is 0.1% !!) 
- how does that affect the total.
- Is there a wiper resistance?
- Is there a endpoint (A, B) minimum resistance?
- you must prototype to learn the details of the devices.

----

> stringing 6 of these 250kohms 

That would make the smallest step 250 ohm, does not meet your requirements.
Furthermore there are only 10k, 50k and 100 k.

So at one end there could be a 10K which has a step size of 10K / 1024 ohm = 9.77 ohm in theory.

### Alternative

An alternative might be [X9C102](https://github.com/RobTillaart/X9C10X)
There is a 1k version with 256 steps so about 3.9 ohm per step.
Then two digipots of 1M (AD5242 ?) which has steps of about 3.9 k.
And one digipot of 100k which has steps of 390 ohm 

This is the smallest set I could find that meet your requirements **in theory**.
There is overlap so it could **in theory** do full range. 
As said before you probably need lookup tables to get the precision / accuracy / repeatability.
And prototype to learn what is possible.

RobTillaart, 2024-12-05:
As this is not an AD524X library specific problem, I propose to open a thread on the Arduino Forum.
There are more people able to help you.

closed 2024-12-05 by RobTillaart