RobTillaart/AD524X #29: Trying to get the AD5242 to work with the Pico

thechordmaster, 2024-12-05:
I am trying this modified code on my Pico 2:

`//
//    FILE: AD524X_isConnected.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: AD524X demo program
//     URL: https://github.com/RobTillaart/AD524X

#include "AD524X.h"

AD5242 AD01(0x2C, &Wire1);     //  AD0 & AD1 == GND

void setup()
{
  
  Serial.begin(115200);
  Serial.println("test");
  Serial.println(__FILE__);
  Serial.print("AD524X_LIB_VERSION: ");
  Serial.println(AD524X_LIB_VERSION);
  Wire.setSDA(16);  //  adjust if needed
  Wire.setSCL(17);
  Wire.begin();

  bool b = AD01.begin();
  Serial.println(b ? "true" : "false");
  Serial.println(AD01.isConnected());

  Serial.println("\nDone...");
}

void loop()
{
}

//  -- END OF FILE --

`

I see no output at the serial monitor..
What could be the issue? i2c checked with scanner for the right address.. 
Vdd and SHdn connected to 3.3v..
ad1, ad0, vis and dgnd are all grounded.

RobTillaart, 2024-12-05:
>  I see no output at the serial monitor..

Try this code

```cpp
#include "AD524X.h"

void setup()
{
  
  Serial.begin(115200);
  Serial.println("test");
  Serial.println(__FILE__);
  Serial.print("AD524X_LIB_VERSION: ");
  Serial.println(AD524X_LIB_VERSION);

  Serial.println("\nDone...");
}

void loop()
{
}

//  -- END OF FILE --
```

does it show output?

thechordmaster, 2024-12-05:
strangely no output! could It be because of an updated board firmware? I use a pico 2 btw.

thechordmaster, 2024-12-05:
Anything you want me to try to help debug?

RobTillaart, 2024-12-05:
If a simple print statement does not work, you need to double check the hardware, wires, etc.

Q: does uploading work?

thechordmaster, 2024-12-05:
Yes uploading works fine… was even able to test out a pwm code without any issues… I tested out the i2c scanner to which I got the address id.. retained same i2c pinout for the digipot as well…

thechordmaster, 2024-12-06:
strange.. this version below works:
void setup()
{
  Serial.begin(9600);
}

void loop()
{
    Serial.println("test");

}

All the default examples work...The println command within the setup is not working

thechordmaster, 2024-12-06:
kindly refer link here:
https://forum.arduino.cc/t/cant-view-serial-print-from-setup/167916

thechordmaster, 2024-12-06:
this code works!
`//
//    FILE: AD524X_isConnected.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: AD524X demo program
//     URL: https://github.com/RobTillaart/AD524X

#include "AD524X.h"

AD524X AD01(0x2C);     //  AD0 & AD1 == GND

void setup()
{
  Serial.begin(115200);
    while (!Serial) ;

  Serial.println();
  Serial.println(__FILE__);
  Serial.print("AD524X_LIB_VERSION: ");
  Serial.println(AD524X_LIB_VERSION);
Wire.setSDA(16);
  Wire.setSCL(17);
  Wire.begin();

  bool b = AD01.begin();
  Serial.println(b ? "true" : "false");
  Serial.println(AD01.isConnected());

  Serial.println("\nDone...");
}

void loop()
{
}

//  -- END OF FILE --

`

RobTillaart, 2024-12-06:
Good to hear you solved it.

So the processor needs more time to initialize the Serial object.
Have seen that before with the Leonardo (?) but did not expect a fast processor like pico would need it.

Not a library problem so no action needed.

thechordmaster, 2024-12-06:
Thank you... I am trying to measure the resistance between Va and Vw, but I see a constant 1.6Mohm reading on my ohmmeter. VDD and SHDN connected to 3.3v.
AD1, AD0, vis and DGND are all grounded.
Is this the right way to check for the code execution? I am running the write to RP2040 code with SDA, SCL pins declared.

RobTillaart, 2024-12-06:
Apparently not.

I have no datasheet nearby to check the pins.  The SHDN pin has the ability to bring the chip to sleep resulting in "frozen" behaviour.

So I would try GND to SHDN and measure again.

thechordmaster, 2024-12-06:
[image]

RobTillaart, 2024-12-06:
Ok, SHDN should be 3.3V if not used. 
So it was wired correctly.

Do you have appropriate pull ups on SDA and SCL? 
For 3.3V you could use 2 kΩ.

thechordmaster, 2024-12-06:
Was able to get it working using this code:
```//
//    FILE: AD5241_write.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: AD5241 demo program - single channel!
//     URL: https://github.com/RobTillaart/AD524X

#include "AD524X.h"

AD5280 AD01(0x2C);  //  AD0 & AD1 == GND

void setup()
{
  Serial.begin(115200);
  while (!Serial) ;

  Serial.println();
  Serial.println(__FILE__);
  Serial.print("AD524X_LIB_VERSION: ");
  Serial.println(AD524X_LIB_VERSION);
Wire.setSDA(16);
  Wire.setSCL(17);

  Wire.begin();
  Wire.setClock(400000);

  bool b = AD01.begin();
  Serial.println(b ? "true" : "false");
  Serial.println(AD01.isConnected());
}

void loop()
{
  
    AD01.write(254);
   delay(10000);
  
}

//  -- END OF FILE --
```

I have completely disconnected A1 from the 3.3 supply. Measuring resistance at A1 and W1.. getting expected values... 4 ohms at minimum position.

Now I would like to increase the range to 2Mohm by connecting W1 to A2 and measuring A2 and W2 after grounding B2... 
In this case  the maximum resistance corresponding to 256 appears as 2 Mohms,, But the minimum resistance is being displayed as 1Mohm, 127- the midpoint shows 1.5MOhms... How can I go about making the span from 0 to 2Mohms?

RobTillaart, 2024-12-06:
So you were using the second I2C bus.
You see hands on prototyping is best learning school.

How to connect, 
Please note I have posted several schemas already.

Connect B1 to A2
Measure R over A1 and B2
Should give about 2 MΩ

 Can you confirm

RobTillaart, 2024-12-06:
Then you can connect W1 to B1
And connect W2 to B2

This will give default 8Ω between A1 and B2

W1 allows to set the first R between 4Ω and 1MΩ
W2 does same for second R.

So when W1 is going from 0 to max, it will change the R between 8 and 1MΩ.

When W1 is at max you can control with W2 from 1 to 2 MΩ.

There are far more combinations possible with W1 and W2, e.g. both halfway also gives about 1 MOhm.

Give it a try.

RobTillaart, 2024-12-06:
Then W1 or W2 is not in the zero position?

RobTillaart, 2024-12-06:
Or maybe both are not in zero pos.

RobTillaart, 2024-12-06:
Do not connect to gnd or 3.3V

thechordmaster, 2024-12-12:
i was not able to get the 8ohms when following the connection you had mentioned. I did get 2Mohm as per your instruction..

I changed the address to 0x2d to activate the second bus…

I am now trying with two separate ad5242 digipots to see if I am able to achieve it.. but I end up getting 1.5mohm stating that one of the resistors is not wiping fully.

I am not sure what the possible solution is

thechordmaster, 2024-12-16:
solution was I had not addressed the second digipot. once done I was able to get the full wiping range.. please close this thread.

closed 2024-12-16 by thechordmaster