RobTillaart/AD524X #32: channels seem to go parallel rather than serial

thechordmaster, 2025-07-02:
Hello

I am having an issue with 4 cascaded digipots where the ad5242 is one of them. 

[image]

I am able to successfully wipe between 80ohms to 1Mohm... But beyond that I am not able to get the second built-in digipot going, After a point, they seem to work in parallel rather than serial resulting in a max output of 1Mohm instead of 2.

I isolated this digipot and in my test, when I remove the A2 from ground, I am able to control both and achieve the full range of 2Mohm. So similarly, I disconnected A2 from ground in my above circuit and tested, but no change. Do you have any advice or pointers to share?

The circuit contains 20 such channels involving multiplexers and shift registers

RobTillaart, 2025-07-02:
Thank you for reporting. Good to see you have succeeded so far.
Currently I am quite busy so I cannot dive into this soon, so only a few questions for now.

Q: which potentiometer(s) in your schema do work (the 80-1 MOhm) correctly?
Q: at what points do you measure?
Q: which potentiometer(s) in your schema do fail?

thechordmaster, 2025-07-02:
> Thank you for reporting. Good to see you have succeeded so far. Currently I am quite busy so I cannot dive into this soon, so only a few questions for now.
> Thank you for your guidance and suggestions. :)
> Q: which potentiometer(s) in your schema do work (the 80-1 MOhm) correctly? 

AD5252 works, AD5245 10k and 100k work as expected.
Q: at what points do you measure? 
measuring between B1 and W2.. A1 and A2 grounded.
Q: which potentiometer(s) in your schema do fail?
AD5242... does not fail.. but acts like its in parallel... so a value of (0,255) and a (1,255) results in 1Mohm between B1 and W2...

Using a PCA 9548A i2c multiplexer and 74HC495 shift registers....

In an isolated setup, upon disconnecting A2 from GND, the AD5242 seems to work across its full span. but when in the circuit refuses to. Pull up resistors required between A1, A2 and Gnd?

RobTillaart, 2025-07-27:
Sorry it took so long, Other priorities take most of my time (coming months)

> After a point, they seem to work in parallel rather than serial resulting in a max output of 1Mohm instead of 2.
In an isolated setup, upon disconnecting A2 from GND, the AD5242 seems to work across its full span

Can indeed be an effect of connecting the A side of the wipers to GND.

You could try to connect the A1 to W1 and A2 to W2 to eliminate any GND effect.
This also prevents them floating and being undefined.

```
CH1 -----[B1   W   A1]----+----- -----[B1   W   A1]----+-----
               ^          |                 ^          |
               +----------+                 +----------+
```

RobTillaart, 2025-08-15:
@thechordmaster 
As you found a solution (as I understand) is this issue still existent?